mbrossar/ai-imu-dr #2: can't download data

StiphyJay, 2019-05-06:
Hello, thanks for your open source code.It's very helpful for me. But i cna't download reformated pickle format of the 00-11 KITTI IMU raw data and training parameters in the link you provided. Would you provide another available link for me? Thank you very much.

mbrossar, 2019-05-06:
Dear StiphyJay,
We can not download data since the server that hosts the data is in maintenance from last friday until to tomorrow evening. Please wait.

mbrossar, 2019-05-07:
The server is now running, you can download data.

closed 2019-05-07 by mbrossar

StiphyJay, 2019-05-07:
Dear StiphyJay:
I still can't download data.Can you upload your data in the Google Cloud.In Cloud Mines Paristech, I have difficulties downloading it.

mbrossar, 2019-05-07:
It is not you that have difficulties, it is the server !

You can download data at https://drive.google.com/file/d/1V8Z2LEu0qDLaxgd0k75R8D_z2OX1Ks8h/view?usp=sharing
Do you need the other files ?

StiphyJay, 2019-05-07:
Yes ,could you give the link about download training parameters? Thank you very much.

mbrossar, 2019-05-07:
This is the link for the training parameters : https://drive.google.com/open?id=11PvRtgY0nLbNza1CdeK0v_hik06aUUSp