mbrossar/ai-imu-dr #3: ModuleNotFoundError: No module named 'liegroups'

Adamquhao, 2019-05-07:
thanks for making your work public, i follow your instructions but something going wrong.....

maosongwang@T480s:~/work/ai-imu-dr/src$ python3 main_kitti.py
IEKF nets loaded
Test filter on sequence: 2011_09_30_drive_0028_extract
Traceback (most recent call last):
  File "main_kitti.py", line 483, in <module>
    launch(KITTIArgs)
  File "main_kitti.py", line 31, in launch
    test_filter(args, dataset)
  File "main_kitti.py", line 437, in test_filter
    to_numpy=True)
  File "/home/maosongwang/work/ai-imu-dr/src/utils.py", line 9, in prepare_data
    t, ang_gt, p_gt, v_gt, u = dataset.get_data(dataset_name)
  File "/home/maosongwang/work/ai-imu-dr/src/dataset.py", line 77, in get_data
    pickle_dict = self[self.datasets.index(i) if type(i) != int else i]
  File "/home/maosongwang/work/ai-imu-dr/src/dataset.py", line 57, in __getitem__
    mondict = self.load(self.path_data_save, self.datasets[i])
  File "/home/maosongwang/work/ai-imu-dr/src/dataset.py", line 148, in load
    pickle_dict = pickle.load(file_pi)
ModuleNotFoundError: No module named 'liegroups'

mbrossar, 2019-05-07:
Can you try it again to download data.zip, I slightly modify the files ?
If you still have this error, you can install the liegroups module, [https://github.com/utiasSTARS/liegroups](https://github.com/utiasSTARS/liegroups)

Adamquhao, 2019-05-07:
thanks for replying, we have addressed this issue by your suggestion. But another issue occurs............when we plot figures, something goes wrong....could u please solve this problem. we couldn't find a proper solution in internet to tackle this.......could u please modify the source code 'utils_plot.py'  in a more general way.

maosongwang@T480s:~/work/ai-imu-dr/src$ python main_kitti.py
IEKF nets loaded
Test filter on sequence: 2011_09_30_drive_0028_extract
Execution time: 12.40 s (sequence time: 538.48 s)
Test filter on sequence: 2011_10_03_drive_0027_extract
Execution time: 10.67 s (sequence time: 471.54 s)
Test filter on sequence: 2011_09_30_drive_0018_extract
Execution time: 6.63 s (sequence time: 288.29 s)
Test filter on sequence: 2011_10_03_drive_0042_extract
Execution time: 2.80 s (sequence time: 121.92 s)
Test filter on sequence: 2011_09_30_drive_0034_extract
Execution time: 2.93 s (sequence time: 127.53 s)
Test filter on sequence: 2011_10_03_drive_0034_extract
Execution time: 10.97 s (sequence time: 484.11 s)
Test filter on sequence: 2011_09_30_drive_0027_extract
Execution time: 2.75 s (sequence time: 115.55 s)
Test filter on sequence: 2011_09_30_drive_0033_extract
Execution time: 3.78 s (sequence time: 165.98 s)
Test filter on sequence: 2011_09_30_drive_0016_extract
Execution time: 0.69 s (sequence time: 29.66 s)
Test filter on sequence: 2011_09_30_drive_0020_extract
Execution time: 2.61 s (sequence time: 115.35 s)
No result for 2011_09_26_drive_0070_extract

Results for: 2011_09_30_drive_0028_extract
Total sequence time: 538.48 s
Traceback (most recent call last):
  File "main_kitti.py", line 483, in <module>
    launch(KITTIArgs)
  File "main_kitti.py", line 34, in launch
    results_filter(args, dataset)
  File "/home/maosongwang/work/ai-imu-dr/src/utils_plot.py", line 204, in results_filter
    fig.savefig(os.path.join(folder_path, fig_name + ".png"))
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/figure.py", line 1834, in savefig
    self.canvas.print_figure(fname, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/backends/backend_qt5agg.py", line 185, in print_figure
    super(FigureCanvasQTAggBase, self).print_figure(*args, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/backend_bases.py", line 2267, in print_figure
    **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/backends/backend_agg.py", line 507, in print_png
    FigureCanvasAgg.draw(self)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/backends/backend_agg.py", line 430, in draw
    self.figure.draw(self.renderer)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/figure.py", line 1299, in draw
    renderer, self, artists, self.suppressComposite)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/image.py", line 138, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/axes/_base.py", line 2437, in draw
    mimage._draw_list_compositing_images(renderer, self, artists)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/image.py", line 138, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/axis.py", line 1147, in draw
    self.label.draw(renderer)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/text.py", line 715, in draw
    bbox, info, descent = textobj._get_layout(renderer)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/text.py", line 308, in _get_layout
    ismath=False)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/backends/backend_agg.py", line 226, in get_text_width_height_descent
    s, fontsize, renderer=self)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/texmanager.py", line 602, in get_text_width_height_descent
    dvifile = self.make_dvi(tex, fontsize)
  File "/home/maosongwang/anaconda3/lib/python3.6/site-packages/matplotlib/texmanager.py", line 393, in make_dvi
    stderr=subprocess.STDOUT)
  File "/home/maosongwang/anaconda3/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/home/maosongwang/anaconda3/lib/python3.6/subprocess.py", line 403, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/home/maosongwang/anaconda3/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/home/maosongwang/anaconda3/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'latex': 'latex'

mbrossar, 2019-05-08:
It seems to be an error with font from latex. I modify the beginning of the file `utils_plot.py`, that is now

```
import matplotlib
import os
from termcolor import cprint
import matplotlib.pyplot as plt
import numpy as np
from itertools import chain
from utils import *
from utils_torch_filter import TORCHIEKF
```

Can you try it ?

It is work, you will probably obtain good results with a possible not so trained filter (flat covariance curve). It is normal, I am waiting for updating the temp.zip file.

mbrossar, 2019-05-09:
Now you can download a trained version of the filter.

closed 2019-05-09 by mbrossar

prayritkhanna, 2019-06-06:
> Can you try it again to download data.zip, I slightly modify the files ?
> If you still have this error, you can install the liegroups module, https://github.com/utiasSTARS/liegroups

Having a little trouble importing  liegroups into the src folder

wronsky317, 2019-10-17:
still have this problem, @mbrossar appreciate!
with data.zip updated, liegroups installed
```
    pickle_dict = pickle.load(file_pi)
ModuleNotFoundError: No module named 'liegroups'
```

by pdb, I find this info:
```
${HOME}/anaconda2/envs/python3/lib/python3.6/site-packages/torch/_utils.py(141)_rebuild_tensor_v2()->tensor([[ 0.0...18,  0.1504]])
-> return tensor
(Pdb) n
--Call--
> <frozen importlib._bootstrap>(966)_find_and_load()
(Pdb) l
[EOF]
(Pdb) n
> <frozen importlib._bootstrap>(968)_find_and_load()
(Pdb) n
> <frozen importlib._bootstrap>(969)_find_and_load()
(Pdb)
> <frozen importlib._bootstrap>(970)_find_and_load()
(Pdb)
> <frozen importlib._bootstrap>(971)_find_and_load()
(Pdb)
ModuleNotFoundError: No module named 'liegroups'
> <frozen importlib._bootstrap>(971)_find_and_load()

```

wronsky317, 2019-10-17:
> still have this problem, @mbrossar appreciate!
> with data.zip updated, liegroups installed
>
> ```
>     pickle_dict = pickle.load(file_pi)
> ModuleNotFoundError: No module named 'liegroups'
> ```
>
> by pdb, I [...]

My fault，ignore this please

restall liegroups solved my trouble

thanks for your sharing!