mbrossar/ai-imu-dr #4: What datasets in KITTI are "00-11 KITTI"?

Beniko95J, 2019-05-24:
Thank you for your great work!

I have some questions about what you mean by "00-11 KITTI". I have not used KITTI dataset before and I see there are multiple categories of datasets e.g. City, Residential and so on, and each category has many sequences. I would like to ask which of them are "00-11 KITTI"?

Thank you very much!

mbrossar, 2019-05-24:
Hi,

You can find this information in the readme of the Kitti odometry development kit:

>
> Mapping to Raw Data
> ===================
> 
> Note that this section is additional to the benchmark, and not required for
> solving the object detection task.
> 
> In order to allow the usage of the laser point clouds, gps data, the right
> camera image and the grayscale images for the TRAINING data as well, we
> provide the mapping of the training set to the raw data of the KITTI dataset.
> The following table lists the name, start and end frame of each sequence that
> has been used to extract the visual odometry / SLAM training set:
> 
> Nr.     Sequence name     Start   End
> ---------------------------------------
> 00: 2011_10_03_drive_0027 000000 004540
> 01: 2011_10_03_drive_0042 000000 001100
> 02: 2011_10_03_drive_0034 000000 004660
> 03: 2011_09_26_drive_0067 000000 000800
> 04: 2011_09_30_drive_0016 000000 000270
> 05: 2011_09_30_drive_0018 000000 002760
> 06: 2011_09_30_drive_0020 000000 001100
> 07: 2011_09_30_drive_0027 000000 001100
> 08: 2011_09_30_drive_0028 001100 005170
> 09: 2011_09_30_drive_0033 000000 001590
> 10: 2011_09_30_drive_0034 000000 001200
> 

The raw sequences can be downloaded from
http://www.cvlibs.net/datasets/kitti/raw_data.php
in the respective category (mostly: Residential).

closed 2019-05-24 by mbrossar

JunsukLee, 2021-08-04:
The ends of the two datasets below have sequential values 33 and 34.
Are sequential datasets like this linked together?

If these data are connected, I would like to increase the recording time by connecting data.

09 : 2011_09_30_drive_0033 000000 001590
10: 2011_09_30_drive_0034 000000 001200

zyqdragon, 2021-10-30:
the information is valuable.

chubukeji, 2022-01-20:
The 03: 2011_09_26_drive_0067 can't be found in official website. Do you have this dataset? Or，do you know why it can't be found? my Email 2661076951@qq.com

YznMur, 2022-05-26:
Hi @chubukeji 
did you find 03: 2011_09_26_drive_0067? I couldn't find it so

chubukeji, 2022-05-26:
@YznMur no, I didn't find it

kefa27, 2022-08-05:
@YznMur, @chubukeji, I can't find it either. Is it already known why it is not there?