mbrossar/ai-imu-dr #5: Question about pickle format

kimjongyoung, 2019-05-29:
I need information about the fields in the pickle files;
'ang_gt'
'name'
'p_gt' 
'sm_gt' 
't' 
't0' 
'u' 
'v_gt'

Thanks in advance.

mbrossar, 2019-05-29:
'ang_gt': ground truth roll, pitch, yaw. size Nx3
'name': name of the sequence
'p_gt': ground truth position. size Nx3
'sm_gt': ground truth zero velocity and zero angular velocity. size Nx2
't': time. size N
't0': initial time
'u': imu inputs (gyro and accelerometer). size Nx6
'v_gt': ground truth velocity. size Nx3

JungHY, 2019-05-30:
what does zero velocity means and also zero angular?
aren't they just zero value?

mbrossar, 2019-06-05:
zero velocity indicates when the vehicle is stopped. It is a one-hot vector. Element 1 means the car is stopping and element 0 the car is moving.
It is the same with  zero angular velocity but for angular velocity.

closed 2019-06-15 by mbrossar