mbrossar/ai-imu-dr #6: Estimation error correction in forward direction

yanming-zou, 2019-06-02:
Hi,

First thanks for your great work and open source.

I have one question in theory. In your algorithm, only the null lateral and vertical velocity assumptions are utilized, but no constraints on the forward direction. I could not understand how the bias in the moving forward direction is estimated and how the estimation error in that direction can be corrected. I mean the state does not violate the 2 assumptions when the errors in that direction are very large. But those errors are not observed in your experiments.

Could you give me some advise about how to explain those phenomenons? 

Thanks a lot.

Yanming

mbrossar, 2019-06-05:
Hi,

Your question is very interesting. As you said, a major question we have to answer is:
"Can we estimate forward accelerometer bias ?"

To answer this question, we can perform an observability analysis (see e.g. this [paper](http://udel.edu/~yuyang/downloads/tr_observability.pdf)). With this analysis, we can see that forward accelerometer bias is observable if the car rotates. My analysis is still a paper draft, I will share you once it is well written.
To understand this good news, we can move from theory to intuition. Assume at time n, you integrate forward bias and the yaw is increasing. At time n+1, a part of the bias is present in the lateral velocity, such that zero lateral velocity measurement gives information about the forward bias value.

yanming-zou, 2019-06-06:
Hi Brossard,

I really appreciated your quick answer. I still have some other questions.

Do you think a litter turn left and then back can correct the estimated bias error? for example changing the lane of your car. Since in your experiments, there are very long straight road. 

Do you think your algorithm can deal with wheeled robot? In my mind, there are at least 2 cases which may not happen for cars: serious slippage (can go forward but not along a straight line) and nearly pure rotation.

Thanks a lot.
Yanming

mbrossar, 2019-06-06:
Hello,

> Do you think a litter turn left and then back can correct the estimated bias error? for example changing the lane of your car. Since in your experiments, there are very long straight road.
It depends on the bias values. If biases are small as in the KITTI dataset, it would work. However, I test the method on another data with important biases (order 10^-3 m/s^2). The position estimates drift in several highway sequences until a 90° curve is performed and then estimation is corrected and accurate for the rest of the sequences. One way to improve the bias estimation is to start by stopping and perform "zero acceleration measurement".

>Do you think your algorithm can deal with wheeled robot? In my mind, there are at least 2 cases which may not happen for cars: serious slippage (can go forward but not along a straight line) and nearly pure rotation.
I assume it will work well for nearly pure rotation since null lateral and vertical velocity assumptions are valid and in this case provide orientation information. 
For serious slippage, the method would presumably badly work. This can be attenuated is the robot is indoor and that its vertical velocity is almost always null.

yanming-zou, 2019-08-02:
Hi Brossard,

I am now interested at the observability analysis of inertial navigation with nonholonomic constraints. I think your new paper is on this topic. So I am really expecting your new paper. Is it ready for publish?
Actually I mainly focus on ground robot with wheel. The 2 main motion patterns of ground robots are: turning with constant speed and moving along a straight line. According to the paper "VINS on Wheels", both of these 2 patterns can add additional unobservable directions to VINS. 
For system with IMU and nonholonomic constraints only (INS), I think turning has no problem. Am I right? In the case of moving along a straight line without rotation, are peach and roll unobservable?
Besides unobservability, the speed of drift (error accumulation) is also an interesting problem. Could we compare the drift speed of VINS and INS with constraints? Do you know if there has been some research in this field?

Best regards
Yanming

mbrossar, 2019-08-19:
Hello,

The paper is currently still in review.

I also think understanding observability/unobservability of a system could help the estimation. As you said, for a wheeled robot, turning is not a problem.
However, in straight line, peach and roll as they can confused with bias.
In my knowledge, there are not so much research in that field. You can look at the other papers of the authors of Vins on wheels, as the paper of the team of Guoquan (Paul) Huang, e.g. Degenerate Motion Analysis for Aided INS With Online Spatial and Temporal Sensor Calibration.

EdernOllivier, 2019-08-21:
Well it is a bit like the bicycle, you need to avance to better feel.

closed 2020-03-12 by mbrossar