mbrossar/ai-imu-dr #14: path_data_base = "./media/mines/46230797-4d43-4860-9b76-ce35e699ea47/KITTI/raw"

gaomomo, 2019-06-11:
Hi,
path_data_base = "./media/mines/46230797-4d43-4860-9b76-ce35e699ea47/KITTI/raw",
I want to know if this path needs to be manually created locally. What data is placed under this path?

Thanks you so much in advance.

prayritkhanna, 2019-06-11:
I have a similar question, when I set the read_data = 1, I get back an error saying there is no such directory: "./media/mines/46230797-4d43-4860-9b76-ce35e699ea47/KITTI/raw". Is there something that I need to do to make read_data work?
[image]

mbrossar, 2019-06-11:
You have to modify this path depending on where you store the KITTI data.

gaomomo, 2019-06-14:
Thank you so much for answering my question.
I want to know what data is placed respectively under these two paths：
./media/mines/46230797-4d43-4860-9b76-ce35e699ea47/KITTI/raw  and   ../data

prayritkhanna, 2019-06-14:
@gaomomo

The first path is the Kitti raw data. This is the data you will be testing/training.

 The "../data" path is where the data should be saved on your system within the ai-imu-dr directory. 
You need to create that directory if you do have one already in there.

gaomomo, 2019-07-04:
Thanks a lot!

closed 2019-07-04 by gaomomo

Akudavale, 2024-01-04:
> You have to modify this path depending on where you store the KITTI data.
from where do i get this KITTI data? 
Thankyou