mbrossar/ai-imu-dr #17: Test doesn't need the trained model?

yzhang93, 2019-07-10:
Hello,

I found your paper very interesting and thanks for sharing the codes! I test the filters by running main_kitti.py, and got similar results as in paper. However, I didn't find in codes where you load the trained model for test. I deleted the provided training parameters (delta_p.p, iekfnets.p, normalize_factors.p), and it still generates the same results. Did you hard coded your trained parameters somewhere?

mbrossar, 2019-07-10:
Hi,

Thanks you for you interrest.
I did not hard coded trained parameters somewhere, they are included in iekfnets.p.

delta_p.p and normalize_factors.p are two files that avoid useless computation. If you delete it, the code will just compute them from data.

If you delete iekfnets.p, you will have a filter without training parameters. The filter still works well but it is better once trained (See Section V-D in the paper). This is particularly remarkable.

yzhang93, 2019-07-10:
@mbrossar Thanks for your reply! 

I'm also wondering how did you set the parameters in class KITTIParameters?
`cov_omega = 2e-4
    cov_acc = 1e-3
    cov_b_omega = 1e-8
    cov_b_acc = 1e-6
    cov_Rot_c_i = 1e-8
    cov_t_c_i = 1e-8
    cov_Rot0 = 1e-6
    cov_v0 = 1e-1
    cov_b_omega0 = 1e-8
    cov_b_acc0 = 1e-3
    cov_Rot_c_i0 = 1e-5
    cov_t_c_i0 = 1e-2
    cov_lat = 1
    cov_up = 10
`
And these values are different compared to those in class Parameters in utils_numpy_filter.py, and also different from the initial values stated in the paper.
`# Process noise covariance
        cov_omega = 1e-3
        """gyro covariance"""
        cov_acc = 1e-2
        """accelerometer covariance"""
        cov_b_omega = 6e-9
        """gyro bias covariance"""
        cov_b_acc = 2e-4
        """accelerometer bias covariance"""
        cov_Rot_c_i = 1e-9
        """car to IMU orientation covariance"""
        cov_t_c_i = 1e-9
        """car to IMU translation covariance"""

        cov_lat = 0.2
        """Zero lateral velocity covariance"""
        cov_up = 300
        """Zero lateral velocity covariance"""

        cov_Rot0 = 1e-3
        """initial pitch and roll covariance"""
        cov_b_omega0 = 6e-3
        """initial gyro bias covariance"""
        cov_b_acc0 = 4e-3
        """initial accelerometer bias covariance"""
        cov_v0 = 1e-1
        """initial velocity covariance"""
        cov_Rot_c_i0 = 1e-6
        """initial car to IMU pitch and roll covariance"""
        cov_t_c_i0 = 5e-3
        """initial car to IMU translation covariance"""`

Xiaojuan6893, 2019-07-19:
Hello, Mr/Ms. Zhang!

Sorry to interrupt, I have tried to download KITTI IMU raw data many times, but I have been unable to succeed. Would you mind sharing it like Baidu Cloud?
 Thank you!

Xiaojuan6893, 2019-07-19:
> Hello,
>
> I found your paper very interesting and thanks for sharing the codes! I test the filters by running main_kitti.py, and got similar results as in paper. However, I didn't find in codes where [...]

Hello, Mr/Ms. Zhang!

Sorry to interrupt, I have tried to download KITTI IMU raw data many times, but I have been unable to succeed. Would you mind sharing it like Baidu Cloud? 
Thank you!

mbrossar, 2019-07-25:
Hello,

You can find my reformatted kitti data in google drive : https://drive.google.com/open?id=1DClhQEDayv2p4IJ1a9XydF-FpnWh5_EQ

You can also find if required temp.zip at https://drive.google.com/open?id=1WPuC71kSb-dq0gSjrrSwWt8WM7XepKmd

Martin
Martin

closed 2019-07-25 by mbrossar