mbrossar/ai-imu-dr #21: Trajectoire avec l'aide d'un ladar

EdernOllivier, 2019-08-02:
Bonjour Martin, pouvez-vous m'en dire plus sur la trajectoire avec l'aide d'un ladar, de quoi s'agit t'il?

mbrossar, 2019-08-02:
Il s'agit de la méthode décrite dans le papier [IMLS-SLAM: scan-to-model matching based on 3D data
](https://arxiv.org/abs/1802.08633)

EdernOllivier, 2019-08-05:
Je n'ai pas fini de regarder cependant je ne vois pas de modèle odometrique, cela doit être normal, dis-moi si il y a une fusion des données, et aussi s'il y a une cartographie embarquée.

EdernOllivier, 2019-08-15:
Bonjour Martin, oui d'accord je vois bien que c'est un algorithme de définition de l'environnement par lidar, cependant ne serait-il pas possible d'associer une cartographie à cette description de l'environnement décrite par un seul capteur, pour faire du SLAM il me semble qu'il est nécessaire d'avoir un modèle odométrique intégré à cette description ?

mbrossar, 2019-08-19:
Bonjour,
Excuse moi, je ne suis pas sur de bien comprendre la question. Le principe du slam consiste à faire simultanément la cartographie et l'odométrie, il y a donc forcément une odométrie.
Si tu parles de modèle d'odométrie "x_{n+1} = f(x_n,...)", non, il n'y en a pas.

EdernOllivier, 2019-08-19:
Bonjour Martin, ma question concerne la mise en relation des points d'intérêt d'une carte avec la description générale et géométrique du Lidar, je me demande comment il est possible de les associer sans un modèle odométrique, pour ce qui concerne la fusion des données d'un véhicule, je crois savoir qu'il y a une indépendance entre les covariances et ce qui permet d'obtenir une estimation assez fine de la position d'un véhicule.

closed 2019-08-19 by EdernOllivier