mbrossar/ai-imu-dr #24: what will happen if this method used in high precision imu navigation?

desa001, 2019-08-17:
Hi,

First thanks for your great work and open source.

I have downloaded your code and run it on my laptop, the result is really cool. But I notice that both the  KITTI dataset and your another article "RINS-W: Robust Inertial Navigation System on Wheels" 's dataset  the IMU gyro and accelermeter precision is really low, it break its limition and reach a good result using iekf and training methond.

Now I am working on high precision IMU navigation such as fog imu(bias stablity < 0.05 deg/h) so I just want to ask : are you think about the proposed method will have good affect on high precision IMU . I want to try, but not for sure.

mbrossar, 2019-08-19:
Hi,

Thanks you for your comments.

If you us an high precision imu, I suggest to first seen the RINS-W paper. With such sensor, the main source of errors is bias. I think the method suits well for correcting this bias when the vehicle is stopped.

EdernOllivier, 2019-08-21:
Well I do suppose that you know the noise and the output frequency of them. If they are faut it is much better for the estimation to be postpone.

EdernOllivier, 2019-08-21:
If they are high

desa001, 2019-08-22:
Hi,

Thanks you for your reply.

In these two articles, the effects of earth rotation and Coriolis acceleration are ignored. However, in the  high precision navigation and long distance navigation (>50km), those effect should not be ignored. In that case, the velocity of the carrier in the inertial frame is equal to $w_{ie}^i \times r^i $ when the carrier stops. Since $r^i $ is unknown, I want to ask how to use ZUPT in this case. Thanks very much.

EdernOllivier, 2019-08-22:
Sure

EdernOllivier, 2019-08-22:
Cf. Les travaux de Faure sur le référentiel inertiel

EdernOllivier, 2019-08-22:
Personnellement, j'avais lu son bouquin "Navigation inertielle et filtrage optimal" et c'est un peu mathématique mais bon on s'habitue à tout

EdernOllivier, 2019-08-23:
Désolé par Pierre Faurre

EdernOllivier, 2019-08-23:
Je viens de le commander et c'est vraiment un bon livre pour mieux comprendre comment cela s'organise et fonctionne en mode inertiel.

closed 2019-12-10 by mbrossar