mbrossar/ai-imu-dr #26: Fusing GPS measurements

amakurin, 2019-08-22:
Hello!
If i want to fuse GPS measurements of position\velocity (which are left invariant in nature) with this RIEKF estimated state should i just do LIEKF propagation\update considering RIEKF state as "previous step" (N-1) state? or there is some special method to do this? Thank you very much for help!

amakurin, 2019-08-26:
I've used method from #22 ignoring "left invariance" aspect of GPS observations, and got acceptable results.

closed 2019-08-26 by amakurin

mbrossar, 2019-08-26:
Hi, 

Sorry for delay. 

I would said something similar as you can add GPS measurements in the right error that loss theoretical left property of left error. But I think it is clearly more simple. 

Martin