mbrossar/ai-imu-dr #28: Huge Position uncertainty

amakurin, 2019-08-26:
Hello again!
Using RIEKF framework (on SE2(3) group as in preliminary paper from RINS-W) i've noticed that with ZUPT, LAT, UP pseudo measurements uncertainty of position reaches huge numbers with order of 1e+5(6..7..). Which can be explained by the fact that we never measure position, but only predict it. But... with time (~20sec with 100hz) this huge numbers give negative influence on filter's stability. Did you experienced this issue in your experiments? or maybe it just caused by my mistakes in filter architecture... Thank you very much for help!

mbrossar, 2019-08-26:
Hi, 

I believe you do not talk about the measurement covariance but about the covariance of the position of the state estimate. 

It is logical to have high values. If you have 10^6 value it's just say one kilometer variance. I have same values but for more than 20 s (frequency does not play a role) . 

Combined with covariance for bias, 10^-9, it can clearly lead to numerical problem. 

I know three solutions. The more simple is to have number in double format, it is what I did in this repo and is sufficient. You can also change the unit in the covariance from meter to km you win a factor 6. The best but more complex in to implement the filter in square root form. Just search square root kalman filter on google. 

Best

Martin

amakurin, 2019-08-27:
Got your point. Thanks!

closed 2019-08-27 by amakurin

EdernOllivier, 2019-08-29:
You need to better initialisé your position and Also that the covariances of measurements should be known.

EdernOllivier, 2019-08-29:
Sorry initialise