mbrossar/ai-imu-dr #29: How could I use my data?

marooncn, 2019-09-03:
Hi, @mbrossar. I want to use my data(rosbag file) to train the network, firstly I need to transfer my data to your input format. The rosbag file has IMU raw data and GPS data, I'm a little confused of the the following pairs:

> (ax, ay, az)  (af, al, au)
> (wx, wy, wz)  (wf, wl, wu)
> (vn, ve, vf) (vf, vl, vu)

What's the difference with them and which one is the sensor raw data?

mbrossar, 2019-09-03:
Each pair corresponds to measurements recorded in a given frame.
I presume you have the same format as Kitti, such that:

vn:    velocity towards north (m/s)
ve:    velocity towards east (m/s)
vf:    forward velocity, i.e. parallel to earth-surface (m/s)
vl:    leftward velocity, i.e. parallel to earth-surface (m/s)
vu:    upward velocity, i.e. perpendicular to earth-surface (m/s)
ax:    acceleration in x, i.e. in direction of vehicle front (m/s^2)
ay:    acceleration in y, i.e. in direction of vehicle left (m/s^2)
az:    acceleration in z, i.e. in direction of vehicle top (m/s^2)
af:    forward acceleration (m/s^2)
al:    leftward acceleration (m/s^2)
au:    upward acceleration (m/s^2)
wx:    angular rate around x (rad/s)
wy:    angular rate around y (rad/s)
wz:    angular rate around z (rad/s)
wf:    angular rate around forward axis (rad/s)
wl:    angular rate around leftward axis (rad/s)
wu:    angular rate around upward axis (rad/s)

You have to use af,al,au, wf,wl,wu, vn, ve,vu.

marooncn, 2019-09-03:
@mbrossar thanks for your reply, do you mean that I should assign the value of raw data of IMU and GPS to _af,al,au, wf,wl,wu, vn, ve,vu_?

marooncn, 2019-09-03:
I pay attention you use _u = [wx, wy, wz, ax, ay, az]_ as IMU input in the code.

marooncn, 2019-09-03:
But why do you take input u = [wx, wy, wz, ax,

marooncn, 2019-09-04:
Hi, I'm still confused. Now I make clear the difference between the variables with subscript (x,y,z) ad (f,l,u). And I need to use the variables with (f,l,u) as your suggestion. But I just read the source code and found that you use (ax, ay, az) and (wx, wy, wz) as input.
```
acc_bis[k, 0] = oxts_k[0].ax
acc_bis[k, 1] = oxts_k[0].ay
acc_bis[k, 2] = oxts_k[0].az
gyro_bis[k, 0] = oxts_k[0].wx
gyro_bis[k, 1] = oxts_k[0].wy
gyro_bis[k, 2] = oxts_k[0].wz
...
u = np.concatenate((gyro_bis, acc_bis), -1)
```

mbrossar, 2019-09-04:
Hi, I remember being myself confused about the data format, and confuse x,y,z indices with n,e,u.
You have to use sensor measurement in the car frame, so x,y,z, it is clear from the KITTI paper 

> Accelerations and angular rates are both specified using two coordinatesystems, one which is attached to the vehicle body(x, y, z)andone that is mapped to the tangent plane of the earth surfaceat that location(f, l, u).

marooncn, 2019-09-04:
Thank you. It works! In the later two days I'll try to train the network with my own data, if there is the result, I'll share with you :)

marooncn, 2019-09-05:
@mbrossar Hi, when I train the model with my own data, there is something wrong:

> File "train_torch_filter.py", line 148, in train_loop
>     loss_train.cuda().backward()
> AttributeError: 'int' object has no attribute 'cuda'

So I cast _loss_train_ to tensor `loss_train=torch.tensor(loss_train)`, and the continue to run, but there is another error:

> RuntimeError: CUDA error: out of memory

actually, the training set only includes six thousand of sequences data and my desktop equips with two nvidia 2080Ti. Why does it?

mbrossar, 2019-09-05:
HI, there are to thinks:

1- Error about loss_train. Lost_train has not to be an integer. Can you check its value ? I presume it would be zero because not updated in the "for i, (dataset_name, Ns)" loop.

2- CUDA error: out of memory. It is normal as you train on six thousand sequences whereas I train on ten sequences. You have to replace the above loop "for i, (dataset_name, Ns)"  by randomly subsampling of very small set (e.g. 10) of your training sequences.

marooncn, 2019-09-05:
Yes, it's because the calculated _loss_ is too large. Its initial value is 402.55727, larger than the _max_loss_(2e1), so it isn't added to _lost_train_, and _lost_train_ keeps 0. I change the _max_loss_ to 2e3 and it works.

marooncn, 2019-09-05:
And the cuda error, when I decrease _seq_dim_ in main_kitti.py. There comes another error:

> File "train_torch_filter.py", line 240, in precompute_lost
>     delta_p = Rot_10_Hz[idxs_0_bis].transpose(-1, -2).matmul(
> RuntimeError: Dimension out of range (expected to be in range of [-1, 0], but got -2)

marooncn, 2019-09-05:
> RuntimeError: Dimension out of range (expected to be in range of [-1, 0], but got -2)

It might be the [pytorch bug](https://github.com/huggingface/pytorch-transformers/issues/175) with python3.5, pytorch1.0 is fixed, but pyorch0.4.1 may not.

marooncn, 2019-09-06:
The training procedure is normal currently. I revise some details as [here](https://github.com/marooncn/ai-imu-dr/commit/86248a5237caa626dace2f4218e14ddf2951d876).

closed 2019-09-06 by marooncn

hmf17, 2022-01-15:
Hi @marooncn , it seems that you have trained and tested with your own IMU data. And I am wondering whether can I ask some questions about the test process.
1, If we use the different IMU data, whether it is necessary to re-train the parameter for the network? Or the pre-trained model can generalize to IMU data from other source?
2, when I test the example I found the pre-train model gived by @mbrossar seems different with the model defined in the utils_torch_filter.py. Because the pre-trained model from the updated url has 5 convolution layers. Could you please share the pre-trained parameters that you download?
Great appreciation!