mbrossar/ai-imu-dr #30: How is the loss calculated?

marooncn, 2019-09-06:
Hi, I know the loss is the the relative error as it's described in the paper:

> We stress the loss function consists of the relative translation error trel

But  how is it calculated in the code?  I'm confused of the following code in _compute_delta_p(Rot, p)_ function. What's the _seq_lengths_? why does it consist of 100, 200, .., 800 ?
```
    seq_lengths = [100, 200, 300, 400, 500, 600, 700, 800]
    k_max = int(Rot.shape[0] / step_size) - 1

    for k in range(0, k_max):
        idx_0 = k * step_size
        for seq_length in seq_lengths:
            if seq_length + distances[idx_0] > distances[-1]:
                continue
            idx_shift = np.searchsorted(distances[idx_0:], distances[idx_0] + seq_length)
            idx_end = idx_0 + idx_shift

            list_rpe[0].append(idx_0)
            list_rpe[1].append(idx_end)

        idxs_0 = list_rpe[0]
        idxs_end = list_rpe[1]
        delta_p = Rot[idxs_0].transpose(-1, -2).matmul(
            ((p[idxs_end] - p[idxs_0]).float()).unsqueeze(-1)).squeeze()
        list_rpe[2] = delta_p
    return list_rpe
```

EdernOllivier, 2019-09-09:
Actually the loss is a much debated issue, because of the mathematical model there is a loss in the computed equations, this should be majorated by the way as the equations are not linear, it depends on the norm taken with the odometric model as you know, please take care of such a linearised odometric model.

marooncn, 2019-09-10:
@EdernOllivier But how can the code make sure the error reasonable? What if my data is 50HZ or even 5HZ?

marooncn, 2019-09-11:
_seq_lengths_ is just to calculate the loss. You can increase or decrease the value according to your data.

closed 2019-09-11 by marooncn