mbrossar/ai-imu-dr #34: The code of another article

Iqun1314, 2019-10-23:
Hi,

First thanks for your great work and open source.

I have downloaded your code and run it on my laptop, the result is really cool. But I notice  your another article "RINS-W: Robust Inertial Navigation System on Wheels" , I have not found the code.
So, if  you can upload the code of the article  "RINS-W: Robust Inertial Navigation System on Wheels", I will be deeply grateful . 

Thank you very much.

last-one, 2019-12-17:
@Iqun1314 Hi, I am re-implementing RINS-W. Do you download the urban dataset and generate the ground-truth of RINS-W? I don't know how to generate the ground-truth of RINS-W by author. But according to my understanding of this article, the statistics I get are very different from the reported by the author. So, I hope to be able to communicate if you are re-implementing RINS-W.

Iqun1314, 2019-12-18:
Ok,I will.
I had implemented another article https://arxiv.org/abs/1904.06064
The urban dataset is contained in the ZIP.
But I don't know how the author extract the dataset from the KITTI dataset.

Iqun1314, 2019-12-18:
您好，很高兴在github上回复我。我是哈尔滨工业大学（深圳）的一名在读研究生，目前在研读RINS-W和AI-IMU这两篇文章，并复现作者的成果。十分希望与您交流。
您能留下联系方式吗？想向您请教关于算法的实现，谢谢！

Send time: Tuesday, Dec 17, 2019 2:33 PM
To: "mbrossar/ai-imu-dr"<ai-imu-dr@noreply.github.com>; 
Cc: "安生"<610181747@qq.com>; "Mention"<mention@noreply.github.com>;

mbrossar, 2020-03-12:
This is the paper of this repo. The code to extract the dataset is based on [pyKitti](https://github.com/utiasSTARS/pykitti).

Regarding the coe of RINS-W, we expect to put it online soonely :)

Iqun1314, 2020-03-13:
> This is the paper of this repo. The code to extract the dataset is based on [pyKitti](https://github.com/utiasSTARS/pykitti).
> 
> Regarding the coe of RINS-W, we expect to put it online soonely :)

Thank you for your reply. I'm looking forword to the code of  RINS-W.

lijiyao111, 2020-08-05:
@mbrossar Hi Martin, I am also very curious to know the details of the implementation of the RINS-W paper. The odometry performance is really good of RINS-W. Following up on the same ask since I saw your reply in March, do you plan to release the code soon in the near future?

Iqun1314, 2020-08-06:
您好，

请问您是从事自动驾驶相关行业的吗？对于RINS-W这篇文章，我也有很过疑惑的地方想与您交流。
能否留个联系方式，比如QQ或者微信。

谢谢：）

兰德智
landz1995@foxmail.com

哈尔滨工业大学（深圳）

mbrossar, 2020-08-06:
I have put the code [here](https://github.com/mbrossar/RINS-W), thank you for recalling me. There are few improvement with the paper, e.g. using CNN rather than LSTM, I let you see.

lijiyao111, 2020-08-06:
@mbrossar Oh, thanks for such a prompt response! That was much quicker than I was expecting. Also thanks for uploading the code! 
Just two quick questions:
In your paper it's using LSTM and in the code I see it's CNN. So you mean that the LSTM network has better performance than CNN?

I see you create the other repo yesterday/today. Seems that the code in that repo is not complete. In one file I see
```
from src.utils import pload, pdump, yload, ydump, mkdir, bmv
from src.utils import bmtm, bmtv, bmmt, pltt, plts, axat, pltt, plts
from src.lie_algebra import SO3, CPUSO3
from src.iekf import RecorderIEKF as IEKF
```
While lie_algebra.py file is there, I did not find iekf.py and utils.py files in the repo. Just to confirm with you.

mbrossar, 2020-08-07:
No, it is the opposite, I mean that CNN network has better performance than LSTM.
I am currently updated the second repo, you can now find iekf.py and utils.py files.

lijiyao111, 2020-08-07:
@mbrossar now I can see all the files, thanks!

Maybe "performance" was not the right word I should use. In the README file, you said the CNN is faster to train than LSTM. I am curious if LSTM has better prediction accuracy than CNN. Could you comment on that from the validation test you did?

mbrossar, 2020-08-19:
I obtained better prediction accuracy with the CNN along with better time execution.

In the readme, I indicate CNN is faster to train than LSTM, which is in fact true. As performance of LSTM and CNN depends on hyperparameter, I can just say that, _on my experiments_, CNN were better.

scott81321, 2021-11-10:
Can you confirm if Brossard's code on:

https://github.com/mbrossar/RINS-W

has the python tools by which to convert a dataset in KITTI format into Brossard's
particular python pickle format ready to be injected into his program in ai-imu-dr  ?

JzHuai0108, 2021-11-24:
@Iqun1314 From your posts, I think you had access to the training data and the trained model. 
Could you please share it with us? 

Apparently, many followers of AI-IMU DR including the author Martin are looking for these data and model.
You may upload it to a cloud service provider like google drive or baidu cloud, and share the link with us.
Thank you so much.

scott81321, 2021-11-24:
I found out that Brossard's code has the means to transform KITTI files into Brossard's pickle files which are composed of 5 python dictionaries apart from time.  It is within main_kitty.py. Now his pickle file data dictionaries are made of u (accel+gyro), and quantities like p_gt which depend on lat/lon , v_gt is velocity Rot_gt are rotation matrices depending on roll, pitch and yaw. These  are indexed and synchronized with the timestamp dictionary 't'.  What I don't know is how GPS outage is modeled.  Brossard's original data could tell us that.  How do you work a test case where you only have u? ie. u and no GPS? For KITTI files, the standard is to replace the last 3 integer numbers with -1, but Brossard's code ignores such numbers.  Can anyone tell me out GPS outage is modeled insofar as the test pickle files are concerned?

mbrossar, 2021-12-19:
Pickle files are now available at new url, please see the [readme](https://github.com/mbrossar/ai-imu-dr)

closed 2021-12-19 by mbrossar