mbrossar/ai-imu-dr #36: AttributeError: Can't get attribute 'SO3'

platohao, 2019-11-05:
Thanks for sharing the code. However, I got the following error:
../data/2011_09_26_drive_0056_extract.p
../data/2011_09_26_drive_0084_extract.p
../data/2011_09_26_drive_0086_extract.p
../data/2011_10_03_drive_0034_extract.p
Traceback (most recent call last):
  File "/home/hao/Desktop/ai-imu-dr/src/main_kitti.py", line 482, in <module>
    dataset = KITTIDataset(args)
  File "/home/hao/Desktop/ai-imu-dr/src/main_kitti.py", line 119, in __init__
    super(KITTIDataset, self).__init__(args)
  File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 56, in __init__
    self.set_normalize_factors()
  File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 96, in set_normalize_factors
    pickle_dict = self.load(self.path_data_save, dataset)
  File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 151, in load
    pickle_dict = pickle.load(file_pi)
AttributeError: Can't get attribute 'SO3' on <module 'liegroups.torch.so3' from '/home/.conda/envs/myenv/lib/python3.6/site-packages/liegroups/torch/so3.py'>

I have installed liegroups package and passed the pytest. Just wonder whether anyone else has encountered the same error before.

Thanks!

fjjjj123e4, 2019-11-16:
I have the same problem. From the official guide of torch, this problem may caused by improper saving method of model. I've not yet solved the problem. 
here's some links may help you and if you make any progress, please contact me. 
https://stackoverflow.com/questions/27732354/unable-to-load-files-using-pickle-and-multiple-modules
https://discuss.pytorch.org/t/error-loading-saved-model/8371

fjjjj123e4, 2019-11-16:
> Thanks for sharing the code. However, I got the following error:
> ../data/2011_09_26_drive_0056_extract.p
> ../data/2011_09_26_drive_0084_extract.p
> ../data/2011_09_26_drive_0086_extract.p
> ../data/2011_10_03_drive_0034_extract.p
> Traceback (most recent call last):
> File "/home/hao/Desktop/ai-imu-dr/src/main_kitti.py", line 482, in
> dataset = KITTIDataset(args)
> File "/home/hao/Desktop/ai-imu-dr/src/main_kitti.py", line 119, in **init**
> super(KITTIDataset, self).**init**(args)
> File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 56, in **init**
> self.set_normalize_factors()
> File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 96, in set_normalize_factors
> pickle_dict = self.load(self.path_data_save, dataset)
> File "/home/hao/Desktop/ai-imu-dr/src/dataset.py", line 151, in load
> pickle_dict = pickle.load(file_pi)
> AttributeError: Can't get attribute 'SO3' on <module 'liegroups.torch.so3' from '/home/.conda/envs/myenv/lib/python3.6/site-packages/liegroups/torch/so3.py'>
> 
> I have installed liegroups package and passed the pytest. Just wonder whether anyone else has encountered the same error before.
> 
> Thanks!

have you solved the problem?

fjjjj123e4, 2019-11-16:
finally i solved the problem
such silly error is caused by the lib 'liesgroup'
the latest version of this package modified the name of SO3 (into SO3matrix), which caused the attribute error
you can git this commit
https://github.com/utiasSTARS/liegroups/tree/01a1554bcc971907bdd3584eed4cd5acee3afc12

platohao, 2019-11-19:
Yes, this solved the problem :)
Thanks a lot for your help!

closed 2019-12-10 by mbrossar