mbrossar/ai-imu-dr #38: About

Iqun1314, 2019-12-04:


Iqun1314, 2019-12-04:
In APPENDIX A , 'As (15) are measurements expressed in the robot’s frame, they lend themselves to the Right IEKF methodology'. But I don't know why it use RIKEF? In other words, when to use LIEKF and when to use RIEKF. Thank you!

mbrossar, 2019-12-10:
You can see both Invariant Kalman Filtering (https://www.annualreviews.org/doi/abs/10.1146/annurev-control-060117-105010), and The Invariant Extended Kalman Filter as a Stable Observer (https://ieeexplore.ieee.org/document/7523335).

RIEKF is adapted for measurements expressed in the robot’s frame, e.g. from camera or zero lateral velocity in **body** frame.
LIEKF is well adapted for measurements expressed in the **global** frame, e.g. GPS.

Iqun1314, 2019-12-18:
Thank you so much！

Iqun1314, 2019-12-18:
Thank you so much！

Send time: Wednesday, Dec 11, 2019 0:05 AM
To: "mbrossar/ai-imu-dr"<ai-imu-dr@noreply.github.com>; 
Cc: "安生"<610181747@qq.com>; "Author"<author@noreply.github.com>;

closed 2020-03-12 by mbrossar

scott81321, 2023-11-07:
I would like to understand this more.  Brossard's paper only mentions RIEKF.  I gather that RIEKF is what is implemented in his code ai-imu-dr.  Is there any way of mixing LIEKF and RIEKF?