mbrossar/ai-imu-dr #43: How to  draw the trajectory from IMU data

Iqun1314, 2020-04-11:
Hello ， in this paper **_AI-IMU Dead-Reckoning_**, the Fig. 1 in this paper.
I want to know how to draw the pure IMU integration (cyan) .
Thank you !

Iqun1314, 2020-06-06:
Hi, I have downloaded the .tex format of the paper .
1- I notice the picture named result on seq. xx of the paper. The trajectory of IMU(cyan) , proposed(green) and so on are all draw by LaTeX tool—tikzpicture. And the data is saved as .txt file. I want to know how to get(calculate) the .txt data, could you tell me the method in detail? 
2- I download reformated pickle format of the 00-11 KITTI IMU raw data of yours. The data as follows:
[image]
Could you tell me these variables represented which value in the paper, thank you so much!
3- And I also want to know the meaning of these variables:
the file _delta_p.p_: **list_rpe_validation**,  **list_rpe**, 
the file _normalize_factors.p_: **num_data**,  **normalize_factors**, **u_std**, **u_loc** . Thank you! I'm looking forward your reply.

mbrossar, 2020-06-07:
The .txt file gather data from input and estimates, and it contains no post-processing.

The variables in pickle data are:
Rot_c_i: rotation between car and IMU
t_c_i: translation between car and IMU
Rot: rotation matrix of IMU
p: position of IMU
b_omega: gyro bias
b_acc: accelerometer bias
t: time
v: velocity of IMU
measurements_covs : the diagonal of cov(y) given by the neural network.

list_rpe and list_rpe_validation contain a list of relation pose error

In normalize_factor, num_date are the number of points, i.e. IMU measurement, of the dataset.
U_std and u_loc in normalize_factors are IMU input standard deviation and mean to learn with normalized data.

Iqun1314, 2020-06-08:
Thank you for such a detailed answer.

Iqun1314, 2020-07-24:
Hi, I have a question of your code:

delta_p = Rot[idxs_0].transpose(-1, -2).matmul(
            ((p[idxs_end] - p[idxs_0]).float()).unsqueeze(-1)).squeeze()

I want to know the formula of delta_p. Could you show me. And I also want to know the meaning of delta_p.

Thank you so much !

mbrossar, 2020-07-26:
Hi,

delta_p is the relative position shift between instants idxs_end and idxs_0. The code is quite strange because it compute a batch of delta_p.

The reason of why having a metric based on delta_p is well explained in the  "A Tutorial on Quantitative Trajectory Evaluationfor  Visual(-Inertial) Odometry" (delta_p is computed in equation (23), paper [link](http://rpg.ifi.uzh.ch/docs/IROS18_Zhang.pdf)).

All the best,

Martin

Iqun1314, 2020-07-26:
Thank you so much ! 

兰德智
landz1995@foxmail.com

哈尔滨工业大学（深圳）

closed 2020-08-19 by mbrossar