mbrossar/ai-imu-dr #47: renormalization of the rotation matrices

shaibagon, 2020-07-26:
In many other works on orientation estimation I noticed there is an additional "re-normalization" step of the rotation matrix, making sure it is unitary and the norm of the columns are all one.

I havn't noticed any such stage in the proposed algorithm.

How can the code guarantee that the estimated rotation matrix, e.g, R_IMU, is indeed a "proper rotation" matrix?

I looked at the resulting "z direction" (last column of the matrix) and noticed that its norm significantly deviates from one.

How can I rectify this?

Thanks,
-Shai

mbrossar, 2020-07-26:
Hello Shai,

The norm of a rotation matrix should be one, but due numerical approximations, it can deviate from one.
To rectify it, you can normalize the rotation matrix.

This is done in the filter [here](https://github.com/mbrossar/ai-imu-dr/blob/master/src/utils_numpy_filter.py#L134) after each "n_normalize_rot" steps.

The function for normalizing rotation matrices is available [here](https://github.com/mbrossar/ai-imu-dr/blob/master/src/utils_numpy_filter.py#L358) for numpy and [here](https://github.com/mbrossar/ai-imu-dr/blob/master/src/utils_torch_filter.py#L411) for pyTorch.

Martin

shaibagon, 2020-07-26:
@mbrossar Thank you very much for the detailed and prompt response! 
Your code is really in very high standard and it is enriching and enlightening to go through it.

closed 2020-07-26 by shaibagon