mbrossar/ai-imu-dr #48: Extending the framework to other types of motion

shaibagon, 2020-07-27:
Hi,
I really like the simplicity of the formulation and the use of "pseudo measurements" to guide the filtering process.

I would like to extend the proposed method to other types of motions, for example a "pendulum like" motion where the IMU frame velocity is mostly in the up/dn direction and zero for lateral /forward motion. Moreover, there is a periodicity to the rotation, velocity and position - they can be assumed as having zero mean.

Is there a way to get a more detailed description of how the measurements Jacobian matrix `H_n` is calculated?

mbrossar, 2020-07-27:
Hi,

I upload some lines about that in a PDF file with LaTeX equations [here](https://github.com/mbrossar/ai-imu-dr/blob/master/temp/sup.pdf). The main idea is to linearize the measurement residual similarly as in a EKF.

Your pendulum like looks interesting, and for me the motion assumption is "no velocity along the direction of the cable" (it may be what you said). 

Do not hesitate if you have question about that, I can extend the document.

Martin

shaibagon, 2020-07-27:
**Thank you!** This is super helpful!

I will have to look into more carefully before I can continue.

Is there a reason why the "clean" acceleration and angular velocity are not part of the "state"?

mbrossar, 2020-07-27:
Acceleration and angular velocity at timestep n are used during propagation at timestep n only.
Adding acceleration and angular velocity in the state would just increase the size of the state error covariance (in that case).

closed 2020-08-06 by mbrossar