mbrossar/ai-imu-dr #50: about the input of he CNN_net

Iqun1314, 2020-10-14:
Hello, I tried to replace the input with wx, wy, wz, ax, ay, az, roll, pitch, yall, the code is:
if iekf.mes_net.training:
        u = dataset.add_noise(u)
        u = u.numpy()     
        u = np.hstack((u, ang_gt))           #6+3=9
        u = torch.from_numpy(u)
but when I run the code, there is an error:
RuntimeError: Given groups=1, weight of size 32 9 5, expected input[1, 6, 2000] to have 9 channels, but got 6 channels instead

I don't know how to deal with it, could you give me some suggestion, thank you!

mbrossar, 2020-10-18:
You give an input of dimension 9 but the network expects an input of dimension 6.
It is sufficient to modify the input dimension (the 6) [here](https://github.com/mbrossar/ai-imu-dr/blob/ddf2a52f758a10f5c20cc1649f747918bade766a/src/utils_torch_filter.py#L47).

scott81321, 2022-11-24:
@Iqun1314 Hello! Have you had success so far?  Can anyone confirm if, whether or not, the training part of ai-imu-dr only considers updating the NN variable of the method?