mbrossar/ai-imu-dr #51: How to get poses from IMU intergreation

jokezhao, 2020-10-20:
Hello ， I want to know how to get poses from the pure IMU integration .It doesn't seem to appear in your paper.Thank you !

mbrossar, 2020-10-21:
The simple way is to run the filter after commenting the update step.

jokezhao, 2020-10-21:
sorry,I don't find the updating step, can you tell me  in detail?

gustavovelascoh, 2021-04-15:
Hi, I think @mbrossar refers to these lines:

https://github.com/mbrossar/ai-imu-dr/blob/ddf2a52f758a10f5c20cc1649f747918bade766a/src/utils_numpy_filter.py#L130

https://github.com/mbrossar/ai-imu-dr/blob/ddf2a52f758a10f5c20cc1649f747918bade766a/src/utils_torch_filter.py#L123

scott81321, 2021-11-21:
Can anyone tell me out GPS outage is modeled insofar as the test pickle files are concerned? Specifically, Brossard's pickle files are made of u (accel+gyro), and quantities like p_gt depend on lat/lon , v_gt is velocity Rot_gt are rotation matrices depending on roll, pitch and yaw. How do you work a test case where you only have u? ie. u and no GPS? For KITTI files, the standard is to replace the last 3 integer numbers with -1, but Brossard's code ignores such numbers.