mbrossar/ai-imu-dr #53: training failed on win10 with GPU or CPU

xxn59, 2020-12-22:
Hi, great job there!
I wished to reproduce the results and encountered some issue.
I changed the value of train_filter to 1 in main_kitti.py and ran the script, but when first epoch loss was calculated, the back propagation stuck for hours and cannot be trained. (in train_torch_filter.py train_loop func)
tried for both GPU and CPU settings and failed the same.
my environment:
CPU: intel 7700
GPU: nvidia rtx2070 w/ cuda 10.1
OS: win10 2004
python3.7 pytorch1.7_gpu

hqtang-git, 2021-07-08:
Hi,
Did you find out the reason, I also take about 434 secs for the first epoch in loss_train.backward() by cpu training model.

Akudavale, 2024-01-17:
@xxn59 any update on ur issues, did you solve it?