mbrossar/ai-imu-dr #54: Confusing about "difference between groundtruth and oxts solution"

LongruiDong, 2020-12-26:
Hi~
Thanks for  your  great open-source work!

When I am reading code , noticed your note [here](https://github.com/mbrossar/ai-imu-dr/blob/ddf2a52f758a10f5c20cc1649f747918bade766a/src/main_kitti.py#L164) ：
```
""" Note on difference between ground truth and oxts solution:
 - orientation is the same
- north and east axis are inverted
- position are closed to but different
=> oxts solution is not loaded
"""
```
What does "ground truth“ and ”oxts solution" specifically refer to here？ 

 In my understanding,  "ground truth" represents the real SE3 (T_w_imu) pose calculated from (lon, lat, alt, roll, pitch yaw),  and Velocity in world frame. When it comes to v_gt, why you use (ve, vn, vu) insted of (vn, ve, vu) ?   

In other words , why need to invert north and east axis？Will it be inconsistent with the coordinates of imu in KITTI?

```
                    v_gt[k, 0] = oxts_k[0].ve
                    v_gt[k, 1] = oxts_k[0].vn
                    v_gt[k, 2] = oxts_k[0].vu
                    v_rob_gt[k, 0] = oxts_k[0].vf
                    v_rob_gt[k, 1] = oxts_k[0].vl
                    v_rob_gt[k, 2] = oxts_k[0].vu
```

I have been troubled by this problem for a long time, and I will be very grateful to you for pointing out my mistake 😁

scott81321, 2021-11-20:
Do you know how to model GPS outage with Brossard's code?

scott81321, 2022-01-28:
Hello @LongruiDong Have you sorted out your questions? Brossard's program seems to be using a mixture of NED and ENU coordinates systems.  The ground velocity (v_gt) seems to be in the ENU format.  However, his position vector is obtain by transforming lat-lon-alt to NED but the he switches N and E while apparently leaving the 3rd coordinate alone and since this 3rd coordinate is in the direction of gravity (yes IMU accel+gyro is forward, leftward and up with gravity recorded as roughly +9.8 m/sec^2 on the sensor).  I for one, would like to know if the RPY is in the NED coordinate system.  I do know his RPY follows 'ZYX' frame format and KITTI's format limits pitch to -pi/2...pi/2 and so I think it follows NED  but I could not find any confirmation of that.