mbrossar/ai-imu-dr #59: how to create  my own  dataset

rainbowrooster, 2021-03-13:
Could you please give me some pointers as to how to prepare your own data (stored in a bag file for instance) in the KITTI format ? Other than the IMU topic what other information or topics should the bag contain ?

scott81321, 2021-11-05:
Hello kahowang, did you ever get a dataset to work on Brossard's program?

kahowang, 2021-11-05:
> Hello kahowang, did you ever get a dataset to work on Brossard's program?

yes，  i had work on this program  through the kitti dataset

scott81321, 2021-11-05:
Thank you for responding!  You see, Brossard's university account was closed because he graduated and moved to industry.  So it's not possible to access his datasets anymore. i.e.:  

https://cloud.mines-paristech.fr/index.php/s/G9frEswwgKHhnr7/download
https://cloud.mines-paristech.fr/index.php/s/XXkh5VanzDW08hD/download
https://cloud.mines-paristech.fr/index.php/s/8YDqD0Y1e6BWzCG

do not work anymore.    

1) Do you have these datasets by any chance?  If so please email them to me at tcscott@gmail.com

2) If not, can you email me a dataset that works with Brossard's program?  (again at tcscott@gmail.com)

-> You see, I have trouble understanding the Kitti format.    As reported by Harish, it seems to have 2 kinds of accel, gyro and velocity data 
(ax, ay, az) (af, al, au)
(wx, wy, wz) (wf, wl, wu)
(vn, ve, vf) (vf, vl, vu)

but I have only one set of 3D IMU accel + gyro data and GPS (lat, lon and alt) at a lower frequency.  Do you know what the redundancy is for>

My email is tcscott@gmail.com if email is easier.  I also have a qq account at 2793247480@qq.com

scott81321, 2021-11-08:
Haven't heard from you.  Anyway, I see the input is necessary for the KITTI format.  One set of Accel+gyro is for the sensor data and the other is the data as viewed from the flat plane of the earth.   What happens if data is missing?  i.e. GPS coordinates are missing as in GPS blackout when traveling inside a tunnel?

scott81321, 2021-11-09:
i tried a KITTI dataset but the program wants a "reformatted pickle format" of KITTI - apparently with a .p file type.  Did ever get Brossard's program to work?

scott81321, 2022-01-24:
Hello @bshmid  Martin Brossard tells me that GPS is only use for error calculations.  The program only needs 3D IMU accel + gyro.    The main_kitti.py shows how to transform from a KITTI format to his pickle format.   His test files are already in pickle format for you. Much has been sorted out at my end.  However, I am slightly confused by some things. His ground translation vector p_gt is in effect lat-long-alt transformed to NED coordinates.  However, he exchanges 1st (N) and 2nd(E) dimensions but leaves the 3rd alone.  He also uses v_gt = [ve,vn,vu] suggesting ENU coordinates. e and n are aligned for p_gt and v_gt but what about the 3rd dimension of p_gt.  He does not negate it.  I know he cancels out gravity.  He has a gravity vector at [0,0,-9.8] but I don't know if he adds or subtracts it from accel.  Question is what coordinate system RPY follow NED? or ENU?

bshmid, 2022-01-25:
Hello @scott81321 
Thanks for your quick responde
Obviously I've got much less understanding in @mbrossar's code however it seams to be that he is only dealing with 2D, well matching the 2 dimentions (E and N) and leaving out the therd one.
As far as datasets being concerned, I was wondering if you can possibly enlighten to me how to feed my '3D IMU accel + gyro' outputs to the program

scott81321, 2022-01-25:
Don't you mean "inputs" to the program?  The program can handle its outputs.  Anyway, the inputs are Brossard's particular pickle datasets i.e. files with type .p only understood in python.

bshmid, 2022-01-25:
Exactly! I meant to asck how to shape out my IMU's outputs to pickle them to be just as good as @mbrossar's ones

bshmid, 2022-01-26:
Hello @scott81321 Sorry for bothering you so much
May be you can just let me know how to contact Martin Brossard
so I can ask him all the questions myself, at the end of the day a program is not worth much if it can work with preset ammount of datasets only.
Appreciate your input.

scott81321, 2022-01-26:
His email is provided by his Github site ie. martin.brossard@mines-paristech.fr
I have to warn you though that he has left University since he graduated and now works in the private sector.  Thus, you would be very lucky to get a response.

ajay1606, 2022-07-08:
Hello @scott81321, Thanks for highlighting this important issue here. Currently, i also tested with the code provided sample data in the .p (pickle format) and everything works fine. 

Would like to know any progress regarding input IMU data through bag file instead .p format!
Are you able to figure it out?

Appreciate any response.

Thanks