mbrossar/ai-imu-dr #63: The data file is not found

devick, 2021-07-12:
The files can't download

https://cloud.mines-paristech.fr/index.php/s/G9frEswwgKHhnr7/download
https://cloud.mines-paristech.fr/index.php/s/XXkh5VanzDW08hD/download
https://cloud.mines-paristech.fr/index.php/s/8YDqD0Y1e6BWzCG

mbrossar, 2021-07-13:
My cloud account of my previous university has been removed. I need to see how I can recover this files.

JinghangLi, 2021-07-21:
Thank you for sharing the code.
Looking forward to your early restoration of the file link and thank you for your work.

mbrossar, 2021-07-21:
Maybe if somebody yet has the files and can send them to me, I can add another link and solve that issue.

improve100, 2021-07-29:
wait for you update new download link.

MrNothing, 2021-08-05:
Would like to have access to the dataset too if possible. Thank you!

KleistvonLiu, 2021-09-07:
Also eager for access to the dataset if possible. Thank you!

lonlonago, 2021-09-29:
need the file too, ths~   send me 394467238@qq.com

scott81321, 2021-11-04:
I would also like these 'download' files.  Without access to these datasets, how can we test ai-imu-dr?

zhangjinglin888, 2021-11-17:
I also need the file too, ths~ send me 2262640702@qq.com

zhangjinglin888, 2021-11-17:
大哥 跪求数据集和训练模型

mbrossar, 2021-12-19:
Pickle files are now available at new url, please see the [readme](https://github.com/mbrossar/ai-imu-dr)

closed 2021-12-19 by mbrossar

scott81321, 2021-12-19:
Cher Martin

Merci bien de la restoration de vos données!

Salutations

Tony Scott