mbrossar/ai-imu-dr #73: about sup.pdf

yuqJin, 2022-03-23:
This is a very good example about solving H, but I would like to ask about this step
[image]
The transpose is gone. And the content in the second bracket does not correspond to formula (5). Can someone explain it to me? thank you very much!

closed 2022-03-23 by yuqJin

Rajat-Arora, 2022-11-06:
Hey @yuqJin were you able to figure it out?

paaraujo, 2022-11-08:
@Rajat-Arora, I hope this can help:

[image]

Rajat-Arora, 2022-11-08:
@paaraujo Thanks a lot for your help. 
Could you please help me derive the complete H matrix of variables corresponding to biases and car to imu rotation and translation?
Specifically, the last 4 terms in the H matrix.
[image]

I would be really grateful for your help.

Rajat-Arora, 2022-11-17:
Hi @paaraujo, could you please describe more about how did you get the negative sign in the bracket in the following equation?
[image]

According to me and following sup.pdf after linearization, it should become,
[image]
and not
[image]

Could you please describe how did you get it?

Also, after getting R, why didn't we take a transpose while substituting it into equation number 6?

linsongxue, 2023-12-06:
> Hi @paaraujo, could you please describe more about how did you get the negative sign in the bracket in the following equation? [image]
>
> According to me and following sup.pdf after linearization, it [...]

Hi @Rajat-Arora, have you solved this question? Would you mind sharing the solution if you have done.