mbrossar/ai-imu-dr #77: Reason for no end frame for some of the sequences

pra-dan, 2023-02-07:
Hi,

I am trying to run training on the kitti dataset by simply changing 
```
        # choose what to do
        read_data = 0
        train_filter = 0
        ...
```
to 
```
        # choose what to do
        read_data = 0
        train_filter = 1
        ...
```

and got an error:
```console
Traceback (most recent call last):
  File "/Users/prashantdandriyal/Desktop/Home/github/selfdrive/ai-imu-dr/src/main_kitti.py", line 483, in <module>
    launch(KITTIArgs)
  File "/Users/prashantdandriyal/Desktop/Home/github/selfdrive/ai-imu-dr/src/main_kitti.py", line 28, in launch
    train_filter(args, dataset)
  File "/Users/prashantdandriyal/Desktop/Home/github/selfdrive/ai-imu-dr/src/train_torch_filter.py", line 61, in train_filter
    prepare_loss_data(args, dataset)
  File "/Users/prashantdandriyal/Desktop/Home/github/selfdrive/ai-imu-dr/src/train_torch_filter.py", line 110, in prepare_loss_data
    Rot_gt = torch.zeros(Ns[1], 3, 3)
TypeError: zeros() received an invalid combination of arguments - got (NoneType, int, int), but expected one of:
 * (tuple of ints size, *, tuple of names names, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
 * (tuple of SymInts size, *, Tensor out, torch.dtype dtype, torch.layout layout, torch.device device, bool pin_memory, bool requires_grad)
```

On tracing it, I noticed that its because `Rot_gt = torch.zeros(Ns[1], 3, 3)` fails for `self.datasets_train_filter["2011_10_03_drive_0042_extract"] = [0, None]` as Ns is populated with None instead of a valid end frame. Is there a reason for keeping this sequence (42) and not mentioning its end frame? I don't think commenting it is a solution here because many other sequences will have to be treated similarly then.

seungjuuuuuu, 2024-07-23:
Hello, I encountered the same problem. How did you finally solve it?

iPAS, 2025-09-25:
I commented the line with 'None' in main_kitti.py

`
...

    def __init__(self, args):
        super(KITTIDataset, self).__init__(args)

        self.datasets_validatation_filter['2011_09_30_drive_0028_extract'] = [11231, 53650]
        # self.datasets_train_filter["2011_10_03_drive_0042_extract"] = [0, None]
        self.datasets_train_filter["2011_09_30_drive_0018_extract"] = [0, 15000]
        # self.datasets_train_filter["2011_09_30_drive_0020_extract"] = [0, None]
        # self.datasets_train_filter["2011_09_30_drive_0027_extract"] = [0, None]
        # self.datasets_train_filter["2011_09_30_drive_0033_extract"] = [0, None]
        self.datasets_train_filter["2011_10_03_drive_0027_extract"] = [0, 18000]
        self.datasets_train_filter["2011_10_03_drive_0034_extract"] = [0, 31000]
        # self.datasets_train_filter["2011_09_30_drive_0034_extract"] = [0, None]
...
`