mbrossar/ai-imu-dr #87: Training Loss of  the network is fluctuating alot.

Akudavale, 2024-01-27:
Hello Everyone , @mbrossar @marooncn 

i have been working on this paper to tuen the network in MesNet() which is used for trainig the noise covarience matric Nn.

Initially i just run the model for 400 epochs without any change in Learning rate and Weightdecay. I stored all the losses calculated by network for each epoch, and lated ploted the graph loss vvs epoch. I have noticed that Loss of the network is fluctuating heavily.

Later I started tuning the network and changed Learning rate and weight decay even introcuced scheduler. But, my effort were in vein as the model's loss kept fluctuating without any significance change from previous publishe values.

I request, if anyone is working on tuning the CNN or optimizingthe CNN. Help me out from this situation.

Thankyou.

Scarleet123, 2025-07-04:
Hello, I encountered the same issue while training the model. I am wondering if you have resolved your issue and would like to consult with you.

Akudavale, 2025-07-04:
Hello,

I faced the same issue but couldn’t resolve the issue.