mbrossar/ai-imu-dr #88: Problem with pre-trained parameters

TiloccaS, 2024-03-28:
hi @mbrossar @marooncn, I think you have done a great job, I wanted to ask if it is possible to have the network weights pre trained updated to the current network configuration, as I have this error:
```

raise RuntimeError('Error(s) in loading state_dict for {}:\n\t{}'.format(
RuntimeError: Error(s) in loading state_dict for TORCHIEKF:
        Unexpected key(s) in state_dict: "mes_net.cov_net.8.weight", "mes_net.cov_net.8.bias", "mes_net.cov_net.12.weight", "mes_net.cov_net.12.bias", "mes_net.cov_net.16.weight", "mes_net.cov_net.16.bias".
        size mismatch for mes_net.cov_net.4.weight: copying a param with shape torch.Size([64, 32, 5]) from checkpoint, the shape in current model is torch.Size([32, 32, 5]).
        size mismatch for mes_net.cov_net.4.bias: copying a param with shape torch.Size([64]) from checkpoint, the shape in current model 
is torch.Size([32])

```

luopengting, 2024-07-07:
Hi, based on the checkpoint provided and my own guesswork, I fixed this problem and verified that the inference results are correct. I hope this will help you.  :D 
code:
https://github.com/luopengting/ai-imu-dr/blob/master/src/utils_torch_filter.py

result:
[image]

devvaibhav455, 2025-07-17:
> > 您好，根据提供的检查点和我的猜测，我修复了这个问题并验证了推理结果是否正确。希望这对你有所帮助。:D代码：https://github.com/luopengting/ai-imu-dr/blob/master/src/utils_torch_filter.py
> > 结果： [image]
> 
> May I ask why you can load it correctly if you change it like this, and humbly ask for advice

[image]

If you take a look at the diff, luopengting has added new layers to match the size of the tensor from the checkpoint with the model itself. It looks like the issue originally happens because either the provided checkpoint is correct but authors forgot to update the model in the code or the other way around. Anyways, I hope it helps. I still need to go through the paper to see if it explains the model to see which is correct. @luopengting @2316219042 Thank you so much for the solution and asking the question respectively.